ivmech/ivport #3: callback event 0x4f525245

kjjordans, 2015-04-01:
I have been trying out my ivPort. I am have tried using your python /
command line method in Example 1 on github. I have the 'A' solder
jumper set on both sides of the board. I keep getting a callback event
0x4f525245. Any ideas what might be causing the error?
I have checked the cables and connectors for continuity.

sgraphic, 2015-04-10:
Bump i have the same issue with mine. I guess its a version 1 since the urgent fix is applied on it connecting pin 11 and 15. When i try capture all the cameras leds a litup. Using your python script.
Im using jumper B on both sides also updated script to 7,15,16. And i also had a run as default A set.
Anything i'm doing wrong?

ivmech, 2015-04-11:
The worst thing that i got same issues from other users via mail and i have noticed that it is caused by chip soldering mistakes by our PCBA services, therefore we have fixed it at revision 2, if you can return your revision 1 board, i can provide you revision 2 board, sorry me for this situation but it is not heavy budget project,

sgraphic, 2015-04-11:
Hello,

That would be great if i could return it and get a new rev 2. 

Of course i understand its not a high budget thing but its a really cool thing. 

What adress do you want me to send this one?

Thanks
Tommy

ivmech, 2015-04-11:
Address is given at www.ivmech.com

closed 2015-04-22 by ivmech