ivmech/ivport #5: mmal: No data received from sensor.HELP!

JasonGN, 2015-06-11:
I have check my three pi-camera all work fine before I attached them to multiplexer . 
First,  I got mmal wrong because I forgot to solder the "3v3" labelled jumper .
Then I Fix it .BUT I got "mmal: No data received from sensor." 
Please help me fix this Wrong..

ivmech, 2015-06-11:
Can you provide me board's photo? There shouldn't be any error if you have red board,,, but please attach photo and let me see board, we may provide you latest board,

JasonGN, 2015-06-11:
[image]
Yes,I am using the RED board and the Rpi2. See~~I run "raspistill" and the camera build-in LED is LIGHTING,so the camera is working but i still recieve the "mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board"

ivmech, 2015-06-11:
Did you check CAM3-CAM4 switch? doesnt all channel work? maybe it had some ESD fault at shipment, also if it doesn't work and you cant fix, let us send you new board,

JasonGN, 2015-06-11:
All channels get the same wrong. Are there all the cameras start capturing the same time when i run raspistill? Can you tell my how to check the board working or not?

AlexVlc, 2015-10-28:
Hi,

I have the same problem.. same board (purchased in september). When I run raspistill or raspivid the program waits some seconds and say "mmal: no data from sensor....", No matter witch camera is selected, when I run raspistill/raspivid the camera one led lights.... 

Can you help me?

Thanks

svenalbert, 2016-02-09:
I'm experiencing exactly the same issues. I tried to measure the data lines and noticed that the lines "CAM_GPIO" and "CAM1_DN1" are not multiplexed. They do not reach the camera at none of the 4 channels.

8bitdee, 2016-06-10:
I have the same problem. I just received my board (Rev. 3) this week. 

Is there a solution for this? I waited a month for the board to arrive and it doesn't work :/

jdaniellhebert, 2017-01-13:
It is very easy to get this error if you have not disabled the other analog muxes.  It is also easy to have an error if the corresponding IIC mux channel is not selected.

Here is some test code that worked for me.  I reverse engineered the boards a bit and have figured out the jumpers. There is working test code too. I got 4 cameras to work but not multiple boards.

https://github.com/jdaniellhebert/ivport_4.