ivmech/ivport #6: mmal: Received unexpected camera control callback event, 0x4f525245

TTerui, 2015-09-15:
I purchased ivPort in order to connect two cameras onto one RaspberryPi2 board.
However, We did not successfully connect them so far. Here is a problem we have.

We connected two cameras into CAM1 and CAM3 ports and tried to run the following sample python program.

-------
import RPi.GPIO as gp
import os
gp.setwarnings(False)
gp.setmode(gp.BOARD)
gp.setup(7, gp.OUT)
gp.setup(11, gp.OUT)
gp.setup(12, gp.OUT)

def main():
　　gp.output(7, False)
　　gp.output(11, False)
　　gp.output(12, True)
　　capture(1)
gp.output(7, False)
　　gp.output(11, True)
　　gp.output(12, False)
　　capture(3)
def capture(cam):
　　cmd = "raspistill -o capture_%d.jpg" % cam
　　os.system(cmd)

if __name__ == "__main__":
　　main()
　　gp.output(7, False)
　　gp.output(11, False)
       gp.output(12, False)

-------

Then we got the following error messsage.
"mmal: Received unexpected camera control callback event, 0x4f525245"

I would appreciate if you could give us your advice on fixing this problem.

Sppyker, 2015-09-17:
Hi, we are also currently attempting to use the same multiplexer and are encountering the same error when attempting to read from CAM1 and CAM3. The specific error message is as follows:

"Received unexpected camera control callback event, 0x%08x" % buf[0].cmd)
picamera.exc.PiCameraRuntimeError: Received unexpected camera control callback event, 0x4f525245"

For our project we are trying to multiplex 4 cameras together, and we found that the CAM2 and CAM4 ports work perfectly fine. This may help you achieve your goal of connecting two cameras to the board, but if you find the solution to why CAM1 and CAM3 are not working it would be great to know.....

I would also appreciate any other help in solving this problem, as we really require all 4 ports working correctly.

eladorb, 2015-11-12:
same here 2 boards (latest rev.3) at first everything was fine
but after booting several times failed to use cam1 and cam3 
i suspect it is very sensitive to either power issues or esd 
tried to use the other one but it failed ootb - again might be esd damage

TTerui, 2015-11-14:
I was examined by all channels but all channels get the same error.
The error I think due to some failures of equipment too.
I will think re-purchase.

cvignac, 2016-03-22:
Currently having the same problem, which appeared after a few booting. The error message is :

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 277, in 'calling callback function'
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 129, in _control_callback
    "Received unexpected camera control callback event, 0x%08x" % buf[0].cmd)
picamera.exc.PiCameraRuntimeError: Received unexpected camera control callback event, 0x4f525245

When testing my multiplexer with the testA from the wiki, I get this message when I start the test, and when I want to take a photo with port 3 or port 4. Port 1 and 2 work well

I checked every camera individually, and they all work well. I have go a 2A power input and nothing more connected to the raspberry, so I don't think power is the problem.

Has anyone managed to overcome this problem ?

jdaniellhebert, 2017-01-13:
I had the same error too.  I tried to follow the example code and it seemed unnecessarily complex with poor documentation of both the code and the hardware...so I followed simplest code example from earlier version of board.

I reverse engineered the boards a bit and have figured out the jumpers. . I got 4 cameras to work but not multiple boards.

https://github.com/jdaniellhebert/ivport_4.

SandBC, 2018-05-24:
Hi! Running into the same issue; did anyone every fix the problem? Thanks!