ivmech/ivport #10: error "mmal_vc_component_enable failed to enable component enospc" using multiplexer

mircopuviani, 2016-02-15:
Hi,
i'm trying to take 4 different pictures using the camera multiplexer and I get the following error:

mmal_vc_component_enable failed to enable component enospc

how can I solve it? Any idea?

Thanks

ivmech, 2016-02-15:
Hi, did you solder 3v3 jumper?

mircopuviani, 2016-02-15:
Hi,
thank you for replying. I've just soldered the 3v3 jumper and now it 
works, I didn't noticed it was not soldered. Thanks again

Best regards

BeppeGitHub, 2016-07-01:
I have the same problem using "ivport dual" shield but with the 3v3 jumper already soldered...

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

Anyone know how to solve this issue?

jacopolazzaro, 2016-11-15:
Same problem as @BeppeGitHub with ivport v2 and Raspberry Pi 3.
I've followed the manual and soldered the headers and the A jumper and tried Example 1.
With only one camera connected (without the multiplexer) everything works perfectly.

Anyone?

jacopolazzaro, 2016-11-17:
IT WORKS!

You have to follow the manual (https://github.com/ivmech/ivport/blob/master/docs/ivport_manual.pdf) for the soldering instructions and this link > https://github.com/ivmech/ivport-v2 for make it run.

Every boot you have to fire init_ivport.py
In the example "test_ivport_quad.py" you have to add the other two cameras to make them shot (around line 49, copy and paste and change the numbers).

With these steps everything is working!