ivmech/ivport #12: Push the settings of the camera, and take picture after

CamAuPoint, 2016-04-17:
Hi,
I just want to know how can we set the exposure_mode to a specific camera, and an other exposure_mode setting to the CAM 2 ?

For example, I got one camera outside and an other one inside my house. The one outside have the sun directly on it, and the one inside my house have no light.

I want to do like a warmup of 10 seconds at the beggining in auto, and after I will take those settings, and apply it to the specific camera that have those settings. So with that, when I switch from CAM1 to CAM2 it wont have to recalculate the exposure mode for about the next 10 minutes.

Is it possible to do that ?

Thanks

jdaniellhebert, 2017-01-13:
To set each camera you have to switch both the MIPI signals (enable and select GPIO) and the IIC mux for each camera.  Once you have switched you can update the camera settings.

I have some test code and have done a bit of reverse engineering that may help with theory of operation.

https://github.com/jdaniellhebert/ivport_4.

I hope that helps.

AlexDupont93, 2017-01-14:
Oh ! Thank you ! When you have done the switch to the camera that you want to update the settings, you send the settings with PiCamera ? Or manually with the I2C code like in the datasheet ?

Thanks !