ivmech/ivport #16: "mmal: no data from sensor...."

8bitdee, 2016-06-11:
This is the message I receive when trying to use the board. I'm using the Revision 3 board and a brand new Raspberry Pi 3. I have all four pi cameras attached properly. 

Any way I can fix this? I spent a lot of money and waited a month for the board to be delivered only to have it not work. I am extremely disappointed and unhappy with this product.

ivmech, 2016-06-13:
Hi, do you use camera module v2 or older version of camera module? ivport is not support camera module v2,

If you use older version of camera module, and error exists, could you share some photos of board and jumper settings and code example that you are using?

markedward94, 2016-07-07:
according to your site, this DOES work with v2 pi cam... if it actually doesn't i will be incredibly disappointed and unhappy as well. 
http://www.ivmech.com/magaza/en/development-modules-c-4/ivport-v2-raspberry-pi-camera-module-v2-multiplexer-p-107

ivmech, 2016-07-18:
ivport v2 support camera module v2,

normal ivport (old version) doesn't support camera module v2, if you
will use camera v2 module, we can supply ivport v2 to you without
additional cost if you ordered after date of ivport v2 was available at
our webpage, please send me your previous order copy,

Thanks,
___
Caner

-- 

Caner DURMUŞOĞLU
Mekatronik Mühendisi
*Ivmech Mekatronik İnovasyon Ltd. Şti.*
Cevizlidere Mah.1208. Cad. No: 4/5
Çankaya / ANKARA
0312 473 00 14
www.ivmech.com <http://www.ivmech.com/>

*Yasal Uyar (TR):*Bu e-posta yalnızca haberleşme amaçlı olarak sadece
ismi yazılı alıcının dikkatine yönelik olarak gönderilmiştir. İsmi
yazılı alıcı olmadığınız takdirde lütfen bu mesajı size gönderen kişiye
iade ediniz ve mesajın aslını imha ediniz. e-postada yer alan şirketi ve
alıcıyı ilgilendiren iş taahhütleri bağlayıcı değildir. Konfirmasyon;
şirketin resmi belgeler yoluyla antetli kağıda yazılmış yetkilinin
imzasının bulunduğu posta ya da faks yoluyla adrese yollanmış onayını
gerektirmektedir. Mesajın ekindeki tüm belgeler virüs kontrolünden
geçirilmiş olup virüslerden kaynaklanabilecek muhtemel zararlar sorumlu
olmayacağımızı ve alıcının belgeyi açmadan önce gerekli kontrolleri
yapmasını tavsiye ederiz.Teşekkür ederiz.
Bu mesajı yazdırmadan önce çevreye verebileceğiniz zararları bir kez
daha düşününüz.
*Disclaimer (EN): *This e-mail hereby was sent to the attention of the
intended recipient for communication purposes only. If you are not the
intended recipient please return this message to the sender and delete
the original message. The business commitments in the e-mail that relate
to the company and the recipient are not binding. The confirmation needs
the approval via official documents duly undersigned by the authorized
officer to be sent to the address by mail or fax on a stationary with
the letterhead. All the documents attached to the message were checked
for viruses; however the company shall not be responsible for any
probable damages due to viruses and we hereby recommend that the
recipient makes necessary controls before opening the document. Thank you.
Think of the environment once more before printing out this message.

jdaniellhebert, 2017-01-13:
I have tested the board with RPI camera 2.  I can get 4x cameras to work on one board.  More than one boards seems to have hardware issue.

https://github.com/jdaniellhebert/ivport_4.