ivmech/ivport #21: IVPORT V2 questions

anfegori91, 2016-09-22:
Hello, I've purchased a IVPORT V2 board and have some questions before attempting to connect the cameras:

1. What are those "X Y" jumpers? I don't see anything related in the documentation.
2. My board came with a little solder blob between R6 and R5 in the place shown in the attached photo. Should I remove it or both leads share that connection?

[image]

Thanks in advance

Longans, 2016-09-29:
Oh, nice catch. Looks like my board has the R5-R6 short issue too. Did you ever get a direct response from the manufacturer?

I'm also confused with some of the jumpers, and the manual hasn't been much help. No idea what X and Y are, and if the GPIO (next to 3V3) needs to be jumper-ed as well.

I did try it out. the Pi can't see the camera with the mux unit in place. I have the two A jumpers shorted and have tried with and without the GPIO pins 7,11, and 12 set for each camera. Never got a connected camera.

Camera and all cables still work after taking the mux board out of the path.

jdaniellhebert, 2017-01-04:
I have 2x boards.  No visible solder balls (if there are they are tiny).  The pads seem connected when I measure impedance.

jdaniellhebert, 2017-01-13:
I reverse engineered the boards a bit and have figured out the jumpers.  There is working test code too. I got 4 cameras to work but not multiple boards.

https://github.com/jdaniellhebert/ivport_4.

bs31, 2017-01-30:
@jdaniellhebert I took a look into your documentation, but couldn't find what X and Y are for. Would you mind to explain that shortly?